chris1111/Wireless-USB-OC-Big-Sur-Adapter #2: Not working in BigSurr beta 1

nikhil19rathore99, 2020-07-09:
I've checked many times but the problem is not in the OC, this USB WiFi works perferctly in macOs 10.15.5 but gives Kernal panic in bigSurr driver used (Wireless USB-OC-Big-Surr-Adapter)
[image]
[image]

Also attached config.plist, please take a look at it

[config.plist.zip](https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/files/4895251/config.plist.zip)

chris1111, 2020-07-09:
I don't have any issue on my Hack HP ProDesk 600 G1 also on my MacBook Pro late 2013 no crash since the first days Big Sur release

nikhil19rathore99, 2020-07-09:
> I don't have any issue on my Hack HP ProDesk 600 G1 also on my MacBook Pro late 2013 no crash since the first days Big Sur release

Then it might the problem from mysids but I've followed as shown in video maybe bigSurr is in Beta that's why it shows

closed 2020-07-13 by chris1111