chris1111/Wireless-USB-OC-Big-Sur-Adapter #4: USB adapter not showed as connecter

adamantinum, 2020-08-08:
Hi,
After I upgraded to Big Sur and installed this new Wireless Utility, I'm not able to connect with my USB adapter (Ralink RT3070), the Utility always show "USB-WiFi not connected". I think the issue is in the kext loading because 
`kmutil showloaded | grep Rt`
and 
cat /var/log/system.log | grep Rt
return anything.
I also tried to manually load kext with
`kmutil load --bundle-path /Volumes/EFI/efi/OC/Kexts/RtWlanU.kext/
kmutil load --bundle-path /Volumes/EFI/efi/OC/Kexts/RtWlanU1827.kext/
kmutil install`
and rebuild cache with
`/usr/bin/kmutil install --volume-root / --check-rebuild`
but it still doesn't work...
Hope it could be an useful feedback.

chris1111, 2020-08-08:
RT3070 is a mediatek Driver, not support here

closed 2020-08-08 by chris1111

adamantinum, 2020-08-08:
Thanks for the reply and sorry for the mistake