chris1111/Wireless-USB-OC-Big-Sur-Adapter #10: USB Wi-Fi Adapter Recognized as Ethernet, not Wi-Fi

arrmo, 2020-11-12:
Hi,

I was able to get the driver working - a bit of manual shuffling, but that's more likely my fault than the driver. What matters though, is that it's working in Big Sur. Thanks!

Just a minor question ... the adapter is being reported / recognized as Ethernet, not as Wi-Fi, so location services isn't happy (or working) on my machine. Is there a configuration setting that needs to be tweaked, to indicate that this is wireless? No biggie if not.

Thanks again.

arrmo, 2020-11-12:
Tried to assign the question label to this, but it seems that I can't. It's not a bug! At least I don't think so 😆

chris1111, 2020-11-12:
This is normal since Catalina

closed 2020-11-12 by chris1111

arrmo, 2020-11-12:
NP, thanks for the info! And again, appreciate the work on this.