chris1111/Wireless-USB-OC-Big-Sur-Adapter #18: Monitor download speed

madvinegar, 2020-12-02:
Is there any way to enable to monitor the download speed? In any network app I try (including the macOs system monitor), the download speed shows as Zero (0).
I thank you for your great work.

ausmbuswinke, 2020-12-02:
Disable SIP still necessary in latest Version?
Thanks for all your work.

Regards
Jürgen

madvinegar, 2020-12-02:
Yes, it is necessary to disable SIP but it is very easy to do with opencore configurator.

chris1111, 2020-12-02:
> Is there any way to enable to monitor the download speed? In any network app I try (including the macOs system monitor), the download speed shows as Zero (0). I am currently in big sur, using latest v.6 of your Big Sur app.
> I thank you for your great work.

You can see on Safari open the Download Arow
[image]

madvinegar, 2020-12-02:
Thank you for your reply. You are correct, but this only applies if I download something via safari. What if I download an OS update (i..e Big Sur 11.0.2 when it is out), or I download an app via the App Store. There is no way to monitor the download speed.
Can this be fixed?

chris1111, 2020-12-02:
You can not find download speed from Appstore like **Installer macOS**  because Apple use /private/var/folders

closed 2020-12-02 by chris1111