chris1111/Wireless-USB-OC-Big-Sur-Adapter #24: OpenCore 0.6.5 not working <- Big Sur

tlacaelelrl, 2021-01-08:
First of all, thank you for the driver.

Works fine with OpenCore version 0.6.1 but when upgrading to 0.6.5 it stops working.
All other hardware works correctly as with 0.6.1 only this wireless adapter is giving me problems.

One more thing:

If I install on version 0.6.5 then attempt to load on 0.6.1 the OS won't boot (loading the OS with version 0.6.5 I can boot), just saying this because I noticed people having issues where they can't boot. Removing the driver then starting and installing on 0.6.1 worked and was able to boot.

chris1111, 2021-01-08:
OpenCore kexts Injection is not my business 😀

closed 2021-01-08 by chris1111

tlacaelelrl, 2021-01-09:
> OpenCore kexts Injection is not my business 😀
> 
> EDIT***
> Use [OpenCore-0.6.6-RELEASE.zip](https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/files/5789754/OpenCore-0.6.6-RELEASE.zip) its work here
> [image]

Well, I actually did not open this thread to blame your driver or OpenCore, I just thought you may want to know your driver is not working (maybe is my system) in that particular version of OpenCore when it was working in previous versions. My system is running, just wanted to update some parts which unfortunately I could not.

Best of luck and hopefully this issue is only on my system or maybe it is just some OpenCore kext injection issue.
By the way, I tried your 0.6.6, did not work either, fortunately as of now I don{t require the OS and also it continues to work under 0.6.1.

P.S. I doubt anyone writing here is trying to blame you (or your work) for anything, I did notice other posts where you say is not your business and in fact, is actually not your business to help others, but I really think (since you decided to go this path) the development communities of this and other subjects are built on feedback which no developer should take personal but rather as help for progress. And maybe the issue is the way we are reporting things, other developers have come around that by pre-asking a set of questions to define whether the issue is in fact their software/script/work or the issue comes from elsewhere. I say this with all respect and just as a thought of how someone from the other side is perceiving your response.

chris1111, 2021-01-09:
You made a big deal out of it for not much I just wanted to show you that it worked for me and OpenCore 0.6.5 also works so the problem comes from your system and I am not here to solve the problem that does not concern the project. It is no sense I would spend my time just on this.

tlacaelelrl, 2021-01-09:
> You made a big deal out of it for not much I just wanted to show you that it worked for me and OpenCore 0.6.5 also works so the problem comes from your system and I am not here to solve the problem that does not concern the project. It is no sense I would spend my time just on this.
> 
> In theory you should have opened a pull request with Acidenthera and they would have asked you for the Logs and all the infos.
> 
> But I confirm that OC 0.6.5 injects the kext

I can also confirm it injects, that does not mean it works.