chris1111/Wireless-USB-OC-Big-Sur-Adapter #29: DWA 131 Big sur OC

Archidas131, 2021-02-07:
hello, 

first of all i would like to thank you for this contribution and the time you spent to adapt the driver, and i hope to find the help i'm seeking , being fairly new to hackintosh.

i have a DWA 131 dongle , on Big Sur OC , SIP is disabled , on legacy boot , i believe that the driver has nothing to do with the EFI .

here is the problem , i followed the video , installed the software and i injected the kexts ( rtwlanu1827.kext rtwlanu.kext) , however when i reboot , a panic shows up and the boot is halted .

i first thought that the kext i have are for clover and not for  OC , i do not know where to download the OC version, 

my config : Latitude 3350
Broadwell-U Intel Core i5-5200U  - 15w
Graphics: Intel HD 5500
LCD: 1366x768
Disk: Toshiba SATA-II 320GB MK3255GSX HDD
Audio: Realtek (alcid=5)
wifi : external dongle DWA 131

thank in advace , and thank you for your efforts , keep up !

chris1111, 2021-02-07:
DWA 131 is not support here this is a old RTL8192SU7 not working after Lion 10.7

closed 2021-02-07 by chris1111

Archidas131, 2021-02-07:
if worked back in catalina just fine

chris1111, 2021-02-07:
> it worked back in catalina just fine

Yes Sorry the V-E  Yes so this is suppose to be work here without any panic
This page is for OC  the Clover version is linked

Archidas131, 2021-02-07:
thanks chris , i use OC , however i didn't understand what i should do to avoid the panic ? , i followed your video step by step , thanks againe

chris1111, 2021-02-07:
I don't know what you should do I have 5 Hackintosh without any issue
and this is test around the world without problems

Archidas131, 2021-02-07:
Allright , thank you , could you please give me a link of your rtwlanu1827.kext rtwlanu.kext ? or where i can find compatibles ones, , i think the versions i have might cause this panic , i'll update you if i fix it by replacing them

chris1111, 2021-02-07:
Unpack the package you can find them

Archidas131, 2021-02-07:
YES , you're awesome bro , i used your Kexts and it works like a charme , you can add DWA 131 as fully working with your driver , really , you're the man