chris1111/Wireless-USB-OC-Big-Sur-Adapter #104: StatusBar IconSet replacement

damiandrake0, 2021-08-31:
Hi Chris, I am using your Wireless-USB-OC-Big-Sur-Adapter and it works very well.
Is it possible to replace the blue icon with one from your zip? If so, how?
I'm waiting for your answer
Damian_Drake

chris1111, 2021-08-31:
Hi @damiandrake0 
Do you have a better Icon Set?

damiandrake0, 2021-09-01:
https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter
At the end of the page it says StatusBar IconSet replacement.zip

chris1111, 2021-09-01:
You can chage the Icon of /Library/Application Support/WLAN
Right clic to StatusBarApp /Show package contents /Resources
Copy and replace all Icons then Logout the mac, after login the Icons wil be show

closed 2021-09-01 by chris1111

damiandrake0, 2021-09-01:
Ok, thank you. I advise you to write it on the page
Bye