chris1111/Wireless-USB-OC-Big-Sur-Adapter #127: My EFI not working after Install v11

mhkanon, 2021-09-28:
I using "Update 15 Sept 2021 Release V11 adapt for latest macOS Big Sur 11" which works fine. But release version "Update 23 Sept 2021 Release V11 correction delete Entry in config if exist" my EFI not working. My macOS not loading after install it. My macOS details are in the attachment
[image]
.

chris1111, 2021-09-28:
If you use update program you aways need to uninstall first

chris1111, 2021-09-28:
EDIT **

Anyway I have atready  test this V11 release before  release it
There is no such a issue with this, I just test now and I show you from picture

**config.plist from Xcode**
[image]

**config.plist from OpenCore Configurator**
[image]

**Boot screen from OpenCore**
[image]

**WIFI working after reboot**
[image]

closed 2021-09-28 by chris1111