chris1111/Wireless-USB-OC-Big-Sur-Adapter #150: SIP for Monterey 12.1 beta

MuttakinHasib, 2021-11-19:
hey, I'm using **Monterey 12.1 beta** & **OC 0.7.5**

### how can I disable SIP?

chris1111, 2021-11-19:
csr-active-config ➤ 67000000 
[image]

closed 2021-11-19 by chris1111

MuttakinHasib, 2021-11-19:
@chris1111 
when I do like this. my PC does not open any application like system preferences, firefox, about this mac, nothing without finder

chris1111, 2021-11-19:
Strange behaviour, you have to investigate?

MuttakinHasib, 2021-11-19:
Is this happening for the beta version ?
Xcode not working, booting time is too long

chris1111, 2021-11-19:
I have the same value on my Main PC 
csr-active-config ➤ 67000000

Everythings work like a charm for 2 years now

MuttakinHasib, 2021-11-20:
[image]

every time it's waiting here for long time then its boot successfully