chris1111/Wireless-USB-OC-Big-Sur-Adapter #156: Cannot Show Your Location

HarryWindsor, 2021-11-30:
When I try to find my location on maps I always get an error: `This Mac cannot determine your current location because it does not have Wi-Fi.`.

I am able to find my WiFi network and connect to it, so this part works fine. Just the location services do not work.

```yaml
802.11n WLAN Adapter:
  Product ID: 0x8176
  Vendor ID: 0x0bda  (Realtek Semiconductor Corp.)
  Version: 2.00
  Serial Number: 00e04c000001
  Speed: Up to 480 Mb/s
  Manufacturer: Realtek
  Location ID: 0x14200000 / 1
  Current Available (mA): 500
  Current Required (mA): 500
  Extra Operating Current (mA): 0
```

Is there something I missed?

[image]

chris1111, 2021-11-30:
Location is part of Icloud (Apple ID), the program as nothing to do with location

closed 2021-11-30 by chris1111

HarryWindsor, 2021-11-30:
So you mean wifi is not the problem? Because macOS tells me that it cannot locate my device because I have no wifi.

[image]

chris1111, 2021-11-30:
Location is not work with this kind of wifi because as you can see the 802,11 WLAN Adapter is detect as Ethernet
For this you need a BCM wifi adapter mac OS compatible