chris1111/Wireless-USB-OC-Big-Sur-Adapter #158: How to install to OpenCore and Big Sur 11.2.1 20D75 if EFI on usb drive?

FanVanDEV, 2021-12-10:
Hello. My EFI located on usb flashcard. I tried to change location in your installer to virtual drive, where installed Hackintosh, and tried to change to my flashcard drive, where located an EFI. After install, I don't see kexts in Kexts folder and in config.plist. What I need to do?

chris1111, 2021-12-10:
if the EFI contain OpenCore and is a mountable partition it will be Mount by the program and install kext in  OC

closed 2021-12-10 by chris1111

FanVanDEV, 2021-12-10:
Where can I download the kexts that your program installs?

chris1111, 2021-12-10:
kexts is Here [EFI.zip](https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/files/7694749/EFI.zip)

FanVanDEV, 2021-12-11:
Thanks! Your kexts work perfectly for me. (I have a small Dexp-adapter, don't know the model)

FREEZX, 2021-12-11:
I have an issue, the installer doesn't seem to install the kexts at the right location for me. I believe it tries to mount an EFI on a wrong drive. However, I managed to install the kexts manually by placing them in the Kexts folder and adding them to my config.plist.
Thanks for the tool!