chris1111/Wireless-USB-OC-Big-Sur-Adapter #214: Searching for WPS (RTL8881ETV)

bahaedin, 2022-03-20:
Hi @chris1111 
I hope you doing well.

Please, I can't close this window its start automatically when I login to my os prevent me from choosing my network manually.

When I click in to the icon in status bar or wireless utility , I receive this window automatically Is there a way to stop it from searching WPS

I have a Chinese USB adapter. **(RTL8188ETV)**

[image]
[image]

chris1111, 2022-03-20:
Never see this?? Check the button on USB adapter if is not jam

bahaedin, 2022-03-20:
There is no button on USB adapter

chris1111, 2022-03-20:
> There is no button on USB adapter

[
](https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/issues/214#issuecomment-1073228531)

bahaedin, 2022-03-20:
> > There is no button on USB adapter
> 
> Check the router

WPS is already disabled on my router, this happens before connecting to wifi, I don't even have access to enter the password, it redirects me directly to this window.

closed 2022-03-29 by chris1111