chris1111/Wireless-USB-OC-Big-Sur-Adapter #218: Wifi connected, but no internet (It works before) (Monterey 12.2)

penguin-io, 2022-03-29:
Hey @chris1111!
TL:DR, so I have been using your wifi app from Catalina to Monterey, and it was working perfectly fine!
But later today, I can't get my internet access, even though the wifi is connected.
Maybe you can help me to solve this problem!
Looking forward to your reply.
Thanks!

P.S. : Here's some screenshots that might be helpful
[image]
[image]

chris1111, 2022-03-29:
from Catalina to Monterey?
if you have update Catalina to Monterey 12 of course you need uninstall, reboot then install again

penguin-io, 2022-03-29:
Hey there @chris1111!
Yeah, I have already reinstall the wifi app at my first copy of Monterey, it's working fine yesterday. Today, the wifi status said connected, but doesn't an internet access. (Have already tried on my phone, It should have an internet access).

chris1111, 2022-03-29:
SSBR, SBR 2 how much SBR you have ?

penguin-io, 2022-03-29:
I have 2 SBR. Purposely made for each floor actually.

chris1111, 2022-03-29:
This is trange because I have 6 Adapter for testing and I never have disconection even after updating macOS since macOS Big sur 11 Beta 1

chris1111, 2022-03-29:
Thats not append on my 6 adapter.
This is as no relation with Wireless-USB-OC-Big-Sur-Adapter, its relate to your side
Issue closed

closed 2022-03-29 by chris1111

penguin-io, 2022-03-30:
Hey there @chris1111!
After looking for several hours of research, found out that it was DHCP router problem.
Thanks @chris1111!