chris1111/Wireless-USB-OC-Big-Sur-Adapter #219: not working in 11.4(ESP OC)

itenl, 2022-04-07:
`USB-WIFI adapter: COMFAST CF-811AC`

`usb-wifi status`
[image]

`csrutil status`

[image]

[step by step the order](https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/issues/168)
in Boot in Recovery
```
csrutil disable
/usr/sbin/spctl kext-consent add ZYM2ETK3E7
```
[image]

chris1111, 2022-04-07:
where you see spctl kext-consent for EFI Partition

chris1111, 2022-04-07:
issue will be delete

closed 2022-04-07 by chris1111