chris1111/Wireless-USB-OC-Big-Sur-Adapter #238: No reason to disable SIP

extremegrief1, 2022-09-02:
Forcing the installer to run on Ventura without the SIP check is working as intended, why do you recommend disabling it and Apple Secure Boot? Also, where is the source code?

chris1111, 2022-09-02:
First of all read the page on GitHub repos before posting issue! you can install SIP enable

closed 2022-09-02 by chris1111

extremegrief1, 2022-09-02:
According to https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/issues/168, you need to first disable SIP, install the package then re-enable it. This is not necessary.

chris1111, 2022-09-02:
There is no way to install kernel extensions on Big Sur and above without SIP disable

chris1111, 2022-09-02:
There is no source for this project, All wifi companies have abandoned it as well as Realtek the main builder.