chris1111/Wireless-USB-OC-Big-Sur-Adapter #239: Kernel panic and stuck on booting

mohammadnt, 2022-09-02:
I followed steps for working with sip enabled, but after installation i cant boot to mac or recovery facing this error:
[image]
Here is my setup:
[image]

chris1111, 2022-09-02:
Make sur in EFI/OC/kexts your not have a duplicate Extensions RtWlanU.kext, RtWlanU1827.kext
You have to remove also from config.plist

mohammadnt, 2022-09-04:
I installed [Wireless-USB-Big-Sur-Adapter](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter) and it worked , then uninstalled it. again i tested oc version and checked for not duplicate items , and it worked! thanks

closed 2022-09-04 by mohammadnt