chris1111/Wireless-USB-OC-Big-Sur-Adapter #247: adaptor take too long to connect or no connection at all

ctf0, 2022-09-11:
Hi there, thank you so much for ur hard work and huge support to the community.

i have `Tp-Link Archer T3U` with `open core 0.8.4` and for some reason the adaptor either take too long to connect like 2-3 mins or no doesnt connect at all.

the strange thing is that i can see all the available networks including the one i automatically connect to, so i know the adaptor is working, also on windows it works perfectly.

so is there anyway i could debug the reason behind that or maybe modify the plist inside the kexts with the correct ids from windows, similar to [USBToolBox](https://github.com/USBToolBox/tool) if any ?

chris1111, 2022-09-11:
Problem comes from your USB setup in OC, I have the same adapter without any issue
Issue closed

closed 2022-09-11 by chris1111

ctf0, 2022-09-11:
> Problem comes from your USB setup in OC

wouldnt this mean that the adaptor doesnt work at all or work very slow ? at least discovering the available networks wont be instant, right ?

chris1111, 2022-09-12:
Tp-Link Archer T3U work good from 10.15 to macOS Ventura 13
I have this adapter for long time and its one of the my preferred
[image]

ctf0, 2022-09-12:
np, thanks again for your help 🥇

chris1111, 2022-09-12:
I can not help here when I know that the problem comes from your config and in no way connect to the adapter and the programs