chris1111/Wireless-USB-OC-Big-Sur-Adapter #248: Source code - Again

maclinit, 2022-09-15:
You mentioned the drivers and USB Adapter app are both original, which is great and we're all able to verify this. 

However, the installer itself, specifically the one that checks if SIP is enabled or not, is made by you and has at least some minimal amount of source code. I'm talking about this here:

[image]

Considering the installer is running with SIP disabled, it would be good for people to be able to see what the installer source is doing, or at the very least write down the specific method you're using to build the installer, even if it's some make-random-installer-studio program. 

The fact that you keep repeating something about these apps being abandoned by Realtek is not a valid excuse to not post the installer source code.

chris1111, 2022-09-15:
I included a script to check the SIP because too many people were using the SIP enable program and posting useless issues like you are doing right now, so you want me to tell you how to bypass this step? The program has been working for three years and more and the SIP checker has been included since Update 15 August 2021
If that doesn't suit you, you just don't have to use it

closed 2022-09-15 by chris1111

maclinit, 2022-09-15:
@chris1111 Since it seems like I wasn't clear in the first post, let me try again in this one.

I don't care at all about bypassing SIP or any other step in the installer. 

What I, and other users want to know is that the installer is safe and has no embedded code to do anything _except_ install wi-fi drivers. 
We don't care how long it's been working, or who abandoned the original project or what useless posts someone is making. 

**We want to make sure this software, which runs without a very important system protection, is safe to install and doesn't contain something that shouldn't be there.**

chris1111, 2022-09-15:
Instead of opening an unnecessary issue; if you want to know if the package is safe why not open it with Suspicious package,  you would have seen that it does not sign and this is the only warning about this program.
Anyway I Will reupload the Package source in the page