chris1111/Wireless-USB-OC-Big-Sur-Adapter #272: No Data received info on Activity Monitor or on iStat Menu

restodo, 2022-12-10:
Hi,
I just updated to the V14 version and it turns up that now I only get Data sent information or upload speed on iStat Menu.
Anyway, thanks for this awesome driver. I am running Big Sur 11.7.1

chris1111, 2022-12-11:
Sorry try to use better thing of  iStat Menu for checking network speed 

Try this [WIFI.Speed.Test.zip](https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/files/10203384/WIFI.Speed.Test.zip)

closed 2022-12-11 by chris1111

restodo, 2022-12-12:
Thanks for the answer. I wasn't talking about speed test, rather to see the download speed. In the below link you can see the download speed in 0 KB/s and it always the same. 

https://share.getcloudapp.com/geuOwWr0

chris1111, 2022-12-12:
Use the App Wifi Speed test then post picture of the screen

restodo, 2022-12-12:
https://share.getcloudapp.com/GGuJokN7

chris1111, 2022-12-12:
you take a screen shot beforre the test finish

restodo, 2022-12-12:
I thought you wanted to see iStat download speed information is in 0 while your test is working.. here is with the test finished.

https://share.getcloudapp.com/YEuDp17l

chris1111, 2022-12-12:
Post only the picture here to your post

restodo, 2022-12-12:
I don't understand what are you asking to do.. regarding the iStat menu, i have to say that it works well with the ethernet and no with your driver. Remember I've told you that with the v12 version it was working well.

chris1111, 2022-12-12:
The Driver as nothing to do with Istat menue

restodo, 2022-12-13:
Thanks for your answer. I've tried other software to check if the problem was iStat and also the download speed isn't show on the bar. This is the software (https://github.com/exelban/stats/).

Check this short video from my screen. Left sensor is from Stat and the other more on the right side is from iStat. In both you can see no download speed information. Stats is free, you can try it to check it on your side.

https://share.getcloudapp.com/L1u6JG6L

restodo, 2022-12-13:
Also i've checked on Activity Monitor and there you can see there isn't Data Received/sec information. This field is always in 0 bytes.