chris1111/Wireless-USB-OC-Big-Sur-Adapter #287: Doesn't boot after Installation in Ventura

agam778, 2023-03-19:
Hey, first of all thanks for the project - I had a spare Archer T4U lying, thought to use that and was able to use it.

My issue now is, after upgrading from macOS Monterey to Ventura v13.2.1, first of all the message showed was that statusbarapp or something similar to that crashed.

Next, I reinstalled the package, it showed the Wifi icon in top bar and also connected.

Now, when I reboot, it is just stuck on Apple Logo, doesn't even show the boot progress bar.

And when I boot without the 2 kexts added by your project, it boots without any issues.

Any kind of help would be appreciated, as I do not have any other choice to use WiFi on my system..Thanks

chris1111, 2023-03-19:
the drivers as not suppose to have incidence on the boot process 
the probleme comes fron your config

closed 2023-03-19 by chris1111

agam778, 2023-03-19:
I don't know, but it's stuck on boot logo on adding those 2 kexts only.

Anyways, I'll see the logs and figure out - something myself then. Thanks for the reply anyways.