chris1111/Wireless-USB-OC-Big-Sur-Adapter #294: Compatibility

AlphaNecron, 2023-05-23:
Is this compatible with T2UB (T2U with Bluetooth)? It's already compatible with T2U but I'm wondering whether it can be used with Bluetooth variant.

chris1111, 2023-05-23:
No Apple Bluetooth is compatible only with Broadcom Bluetooth

closed 2023-05-23 by chris1111

AlphaNecron, 2023-05-24:
Oh, thanks.

mxscowc1ty, 2023-09-19:
> Is this compatible with T2UB (T2U with Bluetooth)? It's already compatible with T2U but I'm wondering whether it can be used with Bluetooth variant.

Bluetooth will work with this USB if you add "BrcmBluetoothInjector" and "BlueToolFixup" kexts, but only with Wi-Fi turned off. I don't know, if it's an application's bug, or the adapter itself, or the system-liminations. 

When Bluetooth is on, you can't work with StatusBarApp anymore — if you'll try, it will break all connections and still show as active, but will stop searching for and connecting to devices.
[image]
[image]

AlphaNecron, 2023-09-19:
Thanks, but I got a AX210, no need this one anymore.

chris1111, 2023-09-19:
> Bluetooth will work with this USB if you add "BrcmBluetoothInjector" and "BlueToolFixup" kexts, but only with Wi-Fi turned off. I don't know, if it's an application's bug (@chris1111 mayb you can give us a hint), or the adapter itself, or the system-liminations.
>

Try with
```
BlueToolFixup.kext
BrcmFirmwareData.kext
BrcmPatchRAM3.kext
```

dont use BrcmBluetoothInjector
?

mxscowc1ty, 2023-09-19:
> Try with
> ```
> BlueToolFixup.kext
> BrcmFirmwareData.kext
> BrcmPatchRAM3.kext
> ```
> 
> dont use BrcmBluetoothInjector ?

Tysm for your reply!

Unfortunately, changed kexts, but the result still the same — Bluetooth and Wi-Fi can't work together, because running the StatusBarApp drops the Bluetooth connection... Perhaps this is an issue with the adapter itself and I'm in any case glad it was able to start up, even though It's not natively supported.

I'm also not demanding that u can fix this, but would be glad if there're any other ideas!

https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/assets/55637151/c1846607-b29b-4905-9f76-20c8e607bb5a

chris1111, 2023-09-19:
> > Try with
> > ```
> > BlueToolFixup.kext
> > BrcmFirmwareData.kext
> > BrcmPatchRAM3.kext
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > dont use BrcmBluetoothInjector ?
> 
> Tysm for your reply!
> 
> Unfortunately, changed kexts, but the result still the same — Bluetooth and Wi-Fi can't work together, because running the StatusBarApp drops the Bluetooth connection... Perhaps this is an issue with the adapter itself and I'm in any case glad it was able to start up, even though It's not natively supported.
> 
> I'm also not demanding that u can fix this, but would be glad if there're any other ideas!
> 
>  Screen.Recording.2023-09-19.at.23-09-50.mp4

It seem your not using latest Utility because the statusBar App is Blue

mxscowc1ty, 2023-09-20:
> It seem your not using latest Utility because the statusBar App is Blue

Oh yeah, hadn't noticed, thanks!

Checked with the latest V17 and the same problem.... If it can't be fixed, I don't insist, it's fine)