chris1111/Wireless-USB-OC-Big-Sur-Adapter #309: Does this driver supports TL-WN722N v3.20?

stereoDotLuigi, 2023-08-07:
I saw tested adapters and I think I will buy
Archer T2U Plus, TL-WN722N or TL-WN825N.

TL-WN722N looked good for me but I have a question.
In TP-Link's website, v3 and v3.20 are different.
And I don't want to buy an adapter that doesn't work with macOS.

I hope you can answer my question. Thank you.

stereoDotLuigi, 2023-08-07:
The answer is actually a big no. It doesn't support because it doesn't have a macOS driver.

closed 2023-08-07 by stereoDotLuigi

chris1111, 2023-08-07:
> I saw tested adapters and I think I will buy
>
> Archer T2U Plus, TL-WN722N or TL-WN825N.
>
>
>
> TL-WN722N looked good for me but I have a question.
>
> In TP-Link's website, v3 and v3.20 are different.
>
> And [...]

V3.2.0 is only for windows

V2 and V3 is good for Mac

stereoDotLuigi, 2023-08-09:
Where can I buy a V2/V3 version of TL-WN722N?
I found this Amazon page, everybody is saying it's V2 or V3.
https://www.amazon.com.tr/TP-Link-TL-WN722N-150Mbps-Kablosuz-Adaptör/dp/9800359850

chris1111, 2023-08-10:
I think its  V2 ?

stereoDotLuigi, 2023-08-11:
I just bought it and it works! Thanks to you!
I think my adapter is V4 because in the box it says "Ver: 4.0"
But the FCC-ID says it's V2. Which one should I trust?

chris1111, 2023-08-11:
See in the  USB 
[image]

stereoDotLuigi, 2023-08-11:
Well, it's weird but look at these:
[image]
[image]

chris1111, 2023-08-11:
see
[image]