chris1111/Wireless-USB-OC-Big-Sur-Adapter #313: Ventura 13.6 not working

pferioli, 2023-09-26:
Hi to all, i've disabled SIP with csrutil disable from secure boot mode and also  executed `sudo spctl --master-disable`i was able to install "Wireless USB Big Sur Adapter-V17" package and i have and icon in the tool bar, but it sais "No Usb-Wifi connection to card", the adapter is a HANA M-C003 adaper based on RTL8188 chipset, any help with be much more appreciated

chris1111, 2023-09-26:
There are a lot of RTL8188 ?
There are many RTL8188 ' What I know is that
 (RTL8188CU) and (RTL8188CUS) only work for a old macOS X Lion 10.7

Ragnarok93, 2023-09-26:
Same here, my adapter worked on 13.5 but is broken on 13.6 even with a fresh install.

chris1111, 2023-09-26:
> Same here, my adapter worked on 13.5 but is broken on 13.6 even with a fresh install. "No USB Wifi Plug In"

All of my adapter work on 13.6 
I have 8 USB Realtek

closed 2023-09-26 by chris1111

Tetoalb, 2023-09-26:
[image0.png]

How do I stop these emails?!

Closed #313<https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/issues/313> as completed.

Ragnarok93, 2023-09-27:
Fixed. User error, forgot to add the kexts from Master