chris1111/Wireless-USB-OC-Big-Sur-Adapter #314: Sonoma iServices weirdness

voidxk, 2023-10-22:
I have ethernet working on my Z390 motherboard but I have a TP-Link Archer T4U V3.20 that I sometimes use and in previous macOS versions everything was working perfectly when using the T4U.

On Sonoma I can no longer receive or send messages or calls, and Messages doesn't sync with my other iDevices. I can sign in and out of iServices/iCloud successfully and other iServices (Notes, Calendar, Contacts) sync fine. As soon as I connect the ethernet Messages syncs again, and I can send and receive messages and calls. Both ethernet and wifi are on the same router/network.

I'm using OC 0.9.4 and I didn't change any kexts or configs since updating to Sonoma. Anyone else experiencing something like this? Probably a Sonoma bug?

danyindrawan, 2023-10-28:
Hi! I thought I was the only one who experienced this. I'm still looking for a solution.
Have you found the way?

voidxk, 2023-10-29:
> Hi! I thought I was the only one who experienced this. I'm still looking for a solution. Have you found the way?

It's an ugly hack but it works if you create new virtual interface and bridge the ethernet interface (even though it's not connected) and the usb wifi interface.

closed 2023-10-29 by chris1111