chris1111/Wireless-USB-OC-Big-Sur-Adapter #344: USB adapter cannot be recognized after being unplugged or computer reboot

TokSakruMai, 2024-06-12:
I tried to install the wireless-usb-oc-big-sur-adapter on my macbookpro 7,1 which I have used open core to update the system to Sonoma(14.5). It does work after I install the adapter. But after I reboot the computer, the usb can no longer be recognized. I have also tried unplugging the usb when it was working normally and when I put it back, it was also unrecognized. Changing the usb port or using the usb hub are both useless.

The only thing I can do to solve this problem is to re-install your driver.

The model of my usb adapter is cf-811ac.

Thank you for any advice!

chris1111, 2024-06-12:
Issue is using OCLP with a snapshot disk
if you solved by reinstall the program thats not an issue

closed 2024-06-12 by chris1111

TokSakruMai, 2024-06-13:
I reset the NVRAM and it works. Thank you very much!!

chris1111, 2024-06-14:
> I reset the NVRAM and it works. Thank you very much!!

Good Idea