chris1111/Wireless-USB-OC-Big-Sur-Adapter #362: <!--

wjimenez27, 2024-10-14:
## 👋 Welcome!
  We’re using Discussions as a place to connect with other members of our community. We hope that you:
  * Ask questions you’re wondering about.
  * Share ideas.
  * Engage with other community members.
  * Welcome others and are open-minded. Remember that this is a community we
  build together 💪.

  To get started, open new discussion with an introduction of yourself and tell us about what you do with this community.
  ## Please don't open discussion for an issue because it would be deleted immediately

_Originally posted by @chris1111 in https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/discussions/19_

wjimenez27, 2024-10-14:
The wireless USB Big Sur app works on my Intel Dell Inspiron i3 3567 7020U CPU Qualcomm Atheros QCA9377 802.11ac Wireless Adapter. The only problem is that it makes my laptop slower, and it freezes, so I have to force restart it. On the other hand, Heliport doesn't freeze my laptop, but the Wi-Fi doesn't work. If you can make a mix (working Wi-Fi from Wireless USB Big Sur + stability from Heliport), then the issues with us (QCA9377) will be fixed. Probably, a more advanced Intel Inspiron, might not get frozen. Please try and advise. @chris1111

chris1111, 2024-10-15:
Its inpossible of freezing with this divers its probably you are mixing with the other thing

chris1111, 2024-10-30:
USB Adapter as no effect on freeze or anything like that.
Issue is on your config trust me.
Close

closed 2024-10-30 by chris1111

wjimenez27, 2024-10-30:
No worries.

I installed Big Sur, and found a video to fix usb, audio, and Bluetooth.

Then I added the ports mapped to use a WiFi dongle.

Finally, I updated until Sequoia, and updated to the latest beta 15.1

No issues found. Every now and then, Bluetooth disconnects, and after I reboot it’s fixed.

See pic attached.

By the way, mine is an i3 Inspiron, not i5. It is just configuration.

Regards,

Lic. Wilson Jimenez

Sent from my iOS cel
________________________________
From: chris1111 ***@***.***>
Sent: Wednesday, October 30, 2024 5:19:54 PM
To: chris1111/Wireless-USB-OC-Big-Sur-Adapter ***@***.***>
Cc: wjimenez27 ***@***.***>; Author ***@***.***>
Subject: Re: [chris1111/Wireless-USB-OC-Big-Sur-Adapter]