chris1111/Wireless-USB-OC-Big-Sur-Adapter #368: Support for the Archer T2UB Nano?

Lit8tech, 2025-01-18:
I understand that the standard Archer T2U Nano is supported by this tool, but what about the [bluetooth version ](https://www.tp-link.com/uk/home-networking/adapter/archer-t2ub-nano/)?
Ofc the bluetooth function will require extra drivers (if those even exist, do you know where I could find them?), but since it's practically the same as the T2U Nano barring the bluetooth, will this tool work with it to get the wifi functionality working on newer macOS versions?

chris1111, 2025-01-18:
Bluetooth on macOS you need the Broadcom Bluetooth, I dont think Archer T2UB is build with Broadcom

Lit8tech, 2025-01-19:
Ah ok, thanks! So if I get the normal T2U nano, this driver and a Broadcom bluetooth adapter [like this one](https://uk.store.asus.com/90ig0070-bw0600-feed-usb-bt400.html) then I don't need any extra stuff for wifi and bluetooth, right?

chris1111, 2025-01-19:
I use this one [kinivo](https://www.kinivo.com/products/kinivo-btd-400-bluetooth-4-0-usb-adapter-for-windows-10-8-7-vista)
With  [BlueToolFixup.kext](https://github.com/acidanthera/BrcmPatchRAM/releases/download/2.6.9/BrcmPatchRAM-2.6.9-RELEASE.zip)

closed 2025-02-04 by chris1111