chris1111/Wireless-USB-OC-Big-Sur-Adapter #369: [Question]: Is it possible to work without "csrutil disable" on Sequoia?

isabsent, 2025-01-20:
I installed the application on **Sequoia**. It works great, thank you! When I **disabled the csrutil**, the application did not load and a message appeared that the application should be deleted. Is this how it should work and work without **csrutil disable** is impossible or did I do something wrong?

chris1111, 2025-01-20:
[Follow the guide](https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/discussions/167)

closed 2025-01-20 by chris1111

isabsent, 2025-01-21:
In your **Guide Readme.pdf** you write:

Double clic Consent.command
Make sur booting OpenCore with
csr-active-config = 03080000
Now Boot in Recovery open Utility / Terminal
Type: the command bellow followed by Enter
/usr/sbin/spctl kext-consent add ZYM2ETK3E7
Closed Terminal then reboot
➤ Followed the guide
Install - Wireless-USB-OC-Big-Sur-Adapter.app ➤ Release ➤ **Release-V16**
Make sur booting OpenCore with
csr-active-config = 00000000
Note: (csrutil disable) (csrutil enable) can be done by the
Bootloader
Follow the Video ⬇

Should I install **Release-V16** which is old (2021) or install newer **Release-V17** for Sequoia (2024)?

chris1111, 2025-01-21:
> In your **Guide Readme.pdf** you write:
> 
> Double clic Consent.command Make sur booting OpenCore with csr-active-config = 03080000 Now Boot in Recovery open Utility / Terminal Type: the command bellow followed by Enter /usr/sbin/spctl kext-consent add ZYM2ETK3E7 Closed Terminal then reboot ➤ Followed the guide Install - Wireless-USB-OC-Big-Sur-Adapter.app ➤ Release ➤ **Release-V16** Make sur booting OpenCore with csr-active-config = 00000000 Note: (csrutil disable) (csrutil enable) can be done by the Bootloader Follow the Video ⬇
> 
> Should I install **Release-V16** (2021) which is old or install **Release-V17** (2024) for Sequoia OpenCore which is newer?

You should install always latest release, I update the guide