chris1111/Wireless-USB-OC-Big-Sur-Adapter #385: Archer T2U Nano not working

5T33Z0, 2025-05-13:
Hi.

I accidentally stumbled over your repo. So I tried the tool in macOS Sonoma 14.7.6 (23H626) with a [TP-Link Archer T2U Nano](https://www.tp-link.com/de/home-networking/adapter/archer-t2u-nano/) Wifi dongle.

After running the installer, the app to connect to wifi hotspots in the menubar stated: "USB Wifi not detected" or something like that. The USB Ports are mapped, the dongle is connected, SIP is lowered (03080000). The kexts were installed and loaded, so I don't know what the issue is. 

Any other things I can try?

Thanks in Advance

chris1111, 2025-05-13:
just try another USB port then reboot

5T33Z0, 2025-05-14:
Thanks. I figured out that the issue is the dongle. There are some contact issue. Wiggling it around made it work. The software and the kexts (current and previous) are working fine.

Thanks for this tool!

closed 2025-05-14 by 5T33Z0